nickolay/esphome-lilygo-t547plus #2: Setting update_interval to never cause reboot loop

lehovec, 2023-02-10:
For some reason, when I set to display component `update_interval` to never, it causes a reboot loop after flash. My config:

```yaml
esphome:
  name: liligo-t5-47-epaper-plus
  platformio_options:
    board_dir: boards
    #include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: lilygo-t5-47-plus

esp32:
  variant: esp32s3
  board: lilygo-t5-47-plus

  framework:
    type: arduino

#.....other conf code

i2c:
  sda: GPIO17
  scl: GPIO18
  scan: false
  id: bus_a

external_components:
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]

text_sensor:
  - platform: homeassistant
    entity_id: weather.forecast_domov
    id: fc_weather
    internal: true
    on_value: # Actions to perform once data for the last sensor has been received
      then:
        - component.update: t547_display

display:
- platform: t547
  id: t5_display
  update_interval: 30s
  lambda: |-
    it.line(0, 0, 960, 540);
```

nickolay, 2023-02-11:
`update_interval: never` works fine for me if I use https://github.com/nickolay/esphome-lilygo-t547plus/blob/main/basic.yaml as base:

```
# The following goes after `wifi: !include wifi-secrets.yaml` from basic.yaml

display:
  - platform: t547
    id: t547_display
    # rotation: 180
    update_interval: never
    lambda: |-
      it.line(0, 0, 960, 540);

i2c:
  sda: GPIO17
  scl: GPIO18
  scan: false
  id: bus_a

external_components:
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]

text_sensor:
  - platform: homeassistant
    entity_id: weather.forecast_home
    id: fc_weather
    internal: true
    on_value: # Actions to perform once data for the last sensor has been received
      then:
        - component.update: t547_display
```

When I try lilygo-t5-47-plus.json without pinning old versions, I do get the loop boot UNLESS I use `fast_connect: true` in the wifi: section. The reason for rebooting is the task watchdog misbehaving:

```
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbmodem14201 with baud rate 115200
[02:19:43][   169][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[02:19:43][   217][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=17 scl=18 freq=100000
[02:19:43][   267][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[02:19:43][   299][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[02:19:43][   299][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[02:19:48]E (5408) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[02:19:48]E (5408) task_wdt:  - IDLE (CPU 0)
[02:19:48]E (5408) task_wdt: Tasks currently running:
[02:19:48]E (5408) task_wdt: CPU 0: loopTask
[02:19:48]E (5408) task_wdt: CPU 1: IDLE
[02:19:48]E (5408) task_wdt: Aborting.
[02:19:48]ESP-ROM:esp32s3-20210327
[02:19:48]Build:Mar 27 2021
[02:19:48]rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[02:19:48]Saved PC:0x40377fa4
WARNING Decoded 0x40377fa4: esp_restart_noos at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)
[02:19:48]SPIWP:0xee
[02:19:48]mode:DIO, clock div:1
[02:19:48]load:0x3fce3808,len:0x43c
[02:19:48]load:0x403c9700,len:0xbec
[02:19:48]load:0x403cc700,len:0x2a3c
[02:19:48]entry 0x403c98d8
[02:19:48][   168][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[02:19:48][   216][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=17 scl=18 freq=100000
[02:19:48][   266][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[02:19:48][   298][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[02:19:48][   299][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[02:19:53]E (5407) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[02:19:53]E (5407) task_wdt:  - IDLE (CPU 0)
[02:19:53]E (5407) task_wdt: Tasks currently running:
[02:19:53]E (5407) task_wdt: CPU 0: loopTask
[02:19:53]E (5407) task_wdt: CPU 1: IDLE
[02:19:53]E (5407) task_wdt: Aborting.
[02:19:53]ESP-ROM:esp32s3-20210327
[02:19:53]Build:Mar 27 2021
[02:19:53]rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[02:19:53]Saved PC:0x40377fa4
...and so on...
```

I don't have much time to debug this further...

nickolay, 2023-12-30:
I finally tracked down the task watchdog crash to `ARDUINO_RUNNING_CORE=0` (https://github.com/esphome/issues/issues/5317), you can try changing it to 1 in the board's json to see if it works.

closed 2023-12-30 by nickolay