nickolay/esphome-lilygo-t547plus #8: ESPHome 2023.12.0 Won't Build

wryandginger, 2023-12-21:
I seem to be getting an error when I try to build with the newest version of ESPHome:

> INFO ESPHome 2023.12.0
INFO Reading configuration /config/esphome/paper1.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'America/Los_Angeles'
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 458, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/data/external_components/319141f9/components/t547/display.py", line 38, in to_code
    await display.register_display(var, config)
  File "/esphome/esphome/components/display/__init__.py", line 119, in register_display
    await cg.register_component(var, config)
  File "/esphome/esphome/cpp_helpers.py", line 56, in register_component
    raise ValueError(
ValueError: Component ID t5_display was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.

bvarick, 2023-12-22:
I think it's an issue with all external display components with the Esphome 2023.12. Here's how it was fixed for a different display:
https://github.com/landonr/lilygo-tdisplays3-esphome/pull/46

wryandginger, 2023-12-22:
Maybe I’m really dense,  but I cannot make heads or tails of that link…it looks like a closed merge. I’m not seeing a fix anywhere. Am I missing something? 

Do I need to make my own fork of this repo and adjust the py files in that merge? Do I delete the external component and just have no display?

bvarick, 2023-12-22:
I haven't figured out a fix yet, I was just posting that to point to how others have fixed the same (at least I think) issue on another display. Perhaps we can do it the same way with this display. Looking at the code that was changed in that PR, it seems it was only a couple of lines in a couple of files.

bvarick, 2023-12-22:
I just forked this repo and made those same changes and its working great on my display. Pull request here: https://github.com/nickolay/esphome-lilygo-t547plus/pull/9

nickolay, 2023-12-22:
@bvarick awesome, thank you!

@wryandginger I merged @bvarick's PR, so it should work for you if you just use `  - source: github://nickolay/esphome-lilygo-t547plus`

closed 2023-12-22 by nickolay