nickolay/esphome-lilygo-t547plus #11: Does using this component cause screen burn in long term?

wasyl00, 2024-03-01:
Hello I just flashed the device using your component and it works great. Trying to evaluate the situation to save myself a headache in the long term. I have seen posts on the HA forums about using esphome and bad/early cases of burn in effect on the T5 screen.

Did you experience any of that? I'm aware of the finite longevity of the device and plan to refresh once in about 10 minutes. How is the quality of the display in your case after a period of time? 

Thx

nickolay, 2024-03-01:
This component uses the library from lilygo to interface with the screen, and the update code is [just a few library calls](https://github.com/nickolay/esphome-lilygo-t547plus/blob/0d8df0dcef48b303e7a87a95060c466bcc593fec/components/t547/t547.cpp#L77), so it shouldn’t cause issues, but my own unit lived through many [experiments](https://github.com/nickolay/esphome-lilygo-t547plus/pull/10#issuecomment-1880049577), so you might want to instead check with someone who has only been using the stable version…

xxLeoxx93, 2025-02-01:
@nickolay I was using it with a refresh every 60sec for a month now and I've pretty obvious burn ins (not sure if permantent or reversable)
I found here that there's a "cycle render" option somewhere but not sure if this could be implemented: https://www.reddit.com/r/homeassistant/comments/sawczc/lilygo_t5_epaper_issuequestion/