nickolay/esphome-lilygo-t547plus #12: Less "aggressive" refresh or partial refresh

Elu43, 2024-08-20:
Hi,

Thank you very much for your work, without you I wouldn't be able to use this screen.
Is there a way to refresh partially ? Or to have a less "aggressive" refresh, when it's blinking 3 times ?

Thanks

nickolay, 2024-08-21:
No. I found that the "aggressive" refresh is [required to avoid damaging the display](https://github.com/nickolay/esphome-lilygo-t547plus/pull/10). The 3 times is [hard-coded by the vendor](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/4a49b32b6db99ca6951d94e0b864f078c20d5432/src/epd_driver.c#L216), presumably they've found fewer cycles do not work as well.

Partial refreshes (i.e. clearing and redrawing a smaller rectangle) are supported by the underlying library, but not implemented in this component.

zk4rz1, 2025-03-30:
I've bought a Lilygo T5 Plus v4.3, and I also had to fight against the unhappy behaviour of the full screen refresh. I think that Lilygo T5 is a very useful display that fits all my needs, so I worked to solve the problem.
I'm not an expert of C++ neither I've ever coded Arduino, but I've found a way to store the color for each pixel modified (in the `void T547::display()` function), verifying this value each time a display update is requested. In this way I can check what are the pixels effectively changed and I can call the base function `epd_clear_area()` only where is really needed.
I've forked the Nickolay Project here: https://github.com/zk4rz1/esphome-lilygo-t547plus-partial-refresh.
Hope this can help someone.