nickolay/esphome-lilygo-t547plus #14: fatal error: sdkconfig.h: No such file or directory

maxPeuck, 2024-09-13:
Hello,

When I try to compile the yaml file with the platformio options, I have the following error : 
`Processing lilygo (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- Wire @ 2.0.0
|-- LilyGoEPD47 @ 0.3.0+sha.4a49b32
Building in release mode
Compiling .pioenvs\lilygo\src\esphome\components\api\api_connection.cpp.o
Compiling .pioenvs\lilygo\src\esphome\components\api\api_frame_helper.cpp.o
Compiling .pioenvs\lilygo\src\esphome\components\api\api_pb2.cpp.o
Compiling .pioenvs\lilygo\src\esphome\components\api\api_pb2_service.cpp.o
<command-line>: warning: "ARDUINO_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_EVENT_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs\lilygo\src\esphome\components\api\api_server.cpp.o
<command-line>: warning: "ARDUINO_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_EVENT_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_EVENT_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs\lilygo\src\esphome\components\api\list_entities.cpp.o
<command-line>: warning: "ARDUINO_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_EVENT_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs\lilygo\src\esphome\components\api\proto.cpp.o
Compiling .pioenvs\lilygo\src\esphome\components\api\subscribe_state.cpp.o
<command-line>: warning: "ARDUINO_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_EVENT_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_EVENT_RUNNING_CORE" redefined
<command-line>: note: this is the location of the previous definition
In file included from c:\users\maxim\.platformio\packages\toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5\xtensa-esp32s3-elf\sys-include\sys\reent.h:503,
                 from C:/Users/maxim/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/newlib/platform_include/sys/reent.h:17,
                 from c:\users\maxim\.platformio\packages\toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5\xtensa-esp32s3-elf\sys-include\math.h:5,
                 from c:\users\maxim\.platformio\packages\toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5\xtensa-esp32s3-elf\include\c++\8.4.0\cmath:45,
                 from src/esphome/core/component.h:3,
                 from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
C:/Users/maxim/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
compilation terminated.`

This is my yaml file:
`esphome:
  name: lilygo
  platformio_options:
    # Unless noted otherwise, based on https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/1eb6119fc31fcff7a6bafecb09f4225313859fc5/examples/demo/platformio.ini#L37
    upload_speed: 921600
    monitor_speed: 115200
    board_build.mcu: esp32s3
    board_build.f_cpu: 240000000L
    board_build.arduino.memory_type: qspi_opi
    board_build.flash_size: 16MB
    board_build.flash_mode: qio
    board_build.flash_type: qspi
    board_build.psram_type: opi
    board_build.memory_type: qspi_opi
    board_build.boot_freq: 80m
    #platform_packages:
    #  - "toolchain-riscv32-esp @8.4.0+2021r2-patch5"
    build_flags:  # the first three defines are required for the screen library to function.
      - "-DBOARD_HAS_PSRAM"
      - "-DARDUINO_RUNNING_CORE=0"  # TODO: this conflicts with the value from platformio's idedata, spewing a lot of warnings during the build.
      - "-DARDUINO_EVENT_RUNNING_CORE=0"  # and this too
      # In addition to lilygo's settings:
      # To enable reading logs over USB until `hardware_uart: USB_CDC` support
      # is added to `logger:`, as detailed in <https://github.com/esphome/feature-requests/issues/1906>:
      - "-DARDUINO_USB_MODE=1"
      - "-DARDUINO_USB_CDC_ON_BOOT=1"

esp32:
  variant: esp32s3
  board: esp32-s3-devkitc-1

external_components:
  - source: github://nickolay/esphome-lilygo-t547plus
    components: ["t547"]

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "BUdpPirnorCZRnfMl+Zt5JZ3TUhkqEEELsaBwcf/wF0="

ota:
  - platform: esphome
    password: "45db4bba5e873f454a8805b91a3edafd"

wifi:
  ssid: xxxx
  password: xxxx

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "kX7nxTB7yrOk"

captive_portal:
  
display:
  - platform: t547
    # rotation: 180
    update_interval: 30s
    lambda: |-
      it.line(0, 0, 960, 540);`

I use esphome 2024.8.3.

Can you help me ?

maxPeuck, 2024-09-13:
the option that cause problem is :
- board_build.arduino.memory_type: qspi_opi

nickolay, 2024-09-13:
The problem is caused by omitting this part: https://github.com/nickolay/esphome-lilygo-t547plus/blob/e7d256b1c5f54458ecc66aed4ef1a1ae513fdb07/basic.yaml#L34-L37

Newer arduino versions [changed `qspi_opi` to `qio_opi`](https://github.com/espressif/arduino-esp32/issues/6958#issuecomment-1179577764), but please use the template that worked for me: https://github.com/nickolay/esphome-lilygo-t547plus/blob/main/basic.yaml if you want something that's known to work.

If you need newer arduino version for some reason, you might want to collaborate with the author of #13, who seems to have found a simpler set of configuration that still works.

closed 2024-09-13 by nickolay

maxPeuck, 2024-09-13:
thank you !
I changed qspi_opi to qio_opi and it works.